Superduper1969/MarstekVenus-LilygoRS485 #9: Template Error in Home assistant (latest version 2025.7.1)

BlueForrest, 2025-07-13:
I'm getting following error in HA:

TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if states('sensor.lilygo_rs485_marstek_ac_power') | float < 0 %} {{ (states('sensor.lilygo_rs485_marstek_ac_power') | float) *-1 }} {% else %} 0 {% endif %}' but no default was specified') while processing template 'Template<template=({% if states('sensor.lilygo_rs485_marstek_ac_power') | float < 0 %} {{ (states('sensor.lilygo_rs485_marstek_ac_power') | float) *-1 }} {% else %} 0 {% endif %}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.my_battery_charging_in_w'

Same for discharge sensor.

error comes from file /homeassistant/packages/marstek_venus_battery_control.yaml

 template:
    - sensor:
        - name: "My Battery Charging in W"
          unique_id: my_battery_charging_in_w
          unit_of_measurement: W
          device_class: power
          state_class: measurement
          state: >
            {% if states('sensor.lilygo_rs485_marstek_ac_power') | float < 0 %}
              {{ (states('sensor.lilygo_rs485_marstek_ac_power') | float) *-1 }}
            {% else %}
              0
            {% endif %}
          
        - name: "My Battery Discharging in W"
          unique_id: my_battery_discharging_in_w
          unit_of_measurement: W
          device_class: power
          state_class: measurement
          state: >
            {% if states('sensor.lilygo_rs485_marstek_ac_power') | float > 0 %}
              {{ (states('sensor.lilygo_rs485_marstek_ac_power') | float) }}
            {% else %}
              0
            {% endif %}

How can I solve this?

Superduper1969, 2025-07-14:
Please check the values of all individual sensors using: 
Developer tools > State > Search/Filter on all individual sensors like "sensor.lilygo_rs485_marstek_ac_power"

BlueForrest, 2025-07-14:
Solved errors by changing 'float'  into  float(0)
just don't know if this is the right solution but so far no more errors.

reinos, 2025-07-16:
I can confirm that `float(0)` does the trick here as well.

closed 2025-07-21 by Superduper1969